atbore-phx/sbam #21: Simpler battery % and time triggers

pwarren, 2024-09-11:
G'day, was looking around for something along these lines for my system, and found a reddit post linking to this code, so thanks!

For my system, I mostly want to prevent grid draw for the morning peak tarrif, and would like to not have to rely on an external API. 
Ideally at ~0500, I'd use the current SOC% and average power draw over the last hour to estimate the percentage at 0700, and if that's going to be > ~18%, set it to charge to that before 0700

I've currently got the system set to charge at 2kw for an hour before 0700, but would be nice to automatically manage whether that happens or not!

From what I could see, there isn't really a way of getting SBAM to do that, I'll have a dive into the code and see what I can do over the next few weeks, but it's been a while since I've done go!

Current plan would be to add another 'mode' like config or estimate, called p_estimate that implements what I want.

Your thoughts?

atbore-phx, 2024-09-12:
hi @pwarren,
yes that's true at this time the only estimate method is by solcast forecast.

Sure you can PR this feature no problem at all and yes it will preferable have a new option to easily integrate the process, i'm glad to have new contributors.

pwarren, 2024-10-31:
Have dived in to fronius modbus doco and found that my system's sunspec mode is set to float, not int+sf, which was causing a lot of headscratching for trying to use sbam!

I've managed to build myself a Python API and some example scripts to do what I want: https://git.pwarren.id.au/pwarren/pyFroniusReg

Will keep plugging way and see if I can modify sbam appropriately

atbore-phx, 2024-11-09:
Hi @pwarren,

Sure, take all the time you need! ;)

Maybe we can add a new parameter to switch between the two modes:

Default: int + sf
Option: float

atbore-phx, 2024-11-09:
Or maybe a fallback method or by detecting the fronius model via the http API ...

Johnnexto, 2024-11-09:
Thanks for good work done!
@pwarren While you are looking at it, may you please consider;
preserve a part of the battery to avoid draining the battery during the expensive hours. (between start_hr and end_hr)
When pw_batt = pw_batt_reserve set MinRsvPct X% further at end_hr set MinRsvPct 5% (where X is function of pw_batt_reserve/battery capacity +5% (or read current MinRsvPct))

My logic is that it is if battery > pw_batt_reserve at start_hr it is cheaper to stop draining of battery rather charge for later use. And release that capacity at end_hr when higher tariff starts.
In addition it would also be very useful to add charging of battery if below MinRsvPct+ pw_batt_reserve (in%) within the start_hr and end_hr. 

Just my thoughts

atbore-phx, 2024-11-10:
Hi @pwarren,

In my inverter, I can choose the Modbus SunSpec type. Both options are available, with int+SF as the default setting.

I'm just curious, why did you set it to float?
What happens if you change it?

[image]

pwarren, 2024-11-11:
@atbore-phx, I didn't set anything in my inverter at first, I think the installer set it to float mode, nothing breaks if I move it to int+SF, and I don't see anything else connecting to it, so maybe that is just the installer's default!

The differences are the register indexes and some register values need to be interpreted as floats, with no scaling factor, but not all.

An illustrative example  is the 'Hz' value from the inverter for the current line frequency
INT: 40086, uint16
 SF: 40087, sunssf
 FLOAT: 40094, float32

so for INT+SF, you get the INT from 40086, multiply by 10 ^ (the INT from 40087)
Float converts directly to the value as a 32bit float.

FLOAT mode still has a bunch of INT+SF registers, things like ChaState (Battery State Of Charge, in %).
INT+SF mode:
ChaState: 40352, uint16 SF: 40366, sunssf
FLOAT mode:
ChaState: 40362, uint16 SF: 40376, sunssf

@Johnnexto 
Yeah, a 'keep a reserve % until the expensive time window' idea sounds like a good idea too, sort of an extension to my idea where we want: "let loads draw from the grid, preserve storage %" instead of my "charge from the grid"

My understanding of the specific 'RsvPct' registers, is that they're about the reserve to keep for backup power, but I've not delved fully in to those yet.

atbore-phx, 2024-11-11:
> I didn't set anything in my inverter at first, I think the installer set it to float mode, nothing breaks if I move it to int+SF, and I don't see anything else connecting to it, so maybe that is just the installer's default!

Hi @pwarren,

Thank you for your work on this issue! After our discussion, it seems that the feature and the float support are quite distinct and could be tackled separately. To make the review process smoother and more manageable, could we focus on the feature in the current MR and address the float support in a subsequent MR?

This way, we can ensure each part gets the attention it deserves without overwhelming the current MR. What do you think?

Thanks again for your contributions!

atbore-phx, 2025-09-14:
hi @pwarren, are you still working on that feature ?

atbore-phx, 2026-01-05:
This issue is closed for inactivity. It can be reopened if needed.