atbore-phx/sbam #22: Suggestions for new functionality

Morten-bigum, 2024-09-22:
Thanks a lot for the time you have put into the development of SBAM. This covers majority of the functionality that I needed to manage my Fronius / BYD solution. To be quite honest I was somewhat surprised that this is not part of the default solution from Fronius, but it is great that you have developed it.

I have two suggestions for enhancement of the functionality:

1) I have two arrays of solar panels on my roof, which face in different directions. Therefore my production estimate is the sum of two Solcast forecasts (one for each array) and this is possible to emulate on the hobbyist account it just requires two calls to Solcast with different ressource_id's and then adding the forecasts. Perhaps it could be feasible to include an URL2 as option setting in the config? Would love to contribute on this, but I'm new to Go :-)

2) I suspect that I later in the year will need to force a charge of my battery during daytime as well. That would be required to ensure I have enough battery capacity to be self-sufficient in the evening peak hours (which are becoming prohibitively expensive). Is this just a matter of extending the crontab hours into the afternoon? Or is there a smarter way?

Thanks a lot. It is really appreciated.
Morten

atbore-phx, 2024-09-24:
Hi @Morten-bigum, 

thanks for using SBAM ;)

about your suggestions:

1. It's a really good feature indeed, and I completely agree with you that we need to support multiple arrays. 
The implementation is not too hard; we just need to create a list for the Solcast API endpoints and iterate the same methods we already have in SBAM, summing the power estimates. 
This feature should be an excellent opportunity to learn golang; the best motivation is coding for our passions, Why don't you give it a try? No deadline here :) . 
I would like to code that in the future, but I can't promise any specific date (maybe in November). I'm quite busy at work at the moment, and SBAM is a hobby for me. Of course, I'm looking for new contributors.

2. Yes, you need to play with crontab and/or start/end hours.
```
-t, --crontab string          crontab (default "0 0 0 0 0")
-e, --end_hr string           END_HR (default "05:55")
-s, --start_hr string         START_HR (default "00:00")
```

atbore-phx, 2024-10-25:
@Morten-bigum, I'll start working on the URL2 option. I think I'll code it for November.

closed 2024-11-09 by atbore-phx