atbore-phx/sbam #25: Trigger setting defaults

andreapalisca, 2024-09-30:
Issue: if HA is not running (crashed or power outage) at the end of the favourable time window and the inverter was charging the battery, the inverter keeps charging until the start of the next favourable window, which can be a day later. I'm running HAOS and I haven't found a way to manually set defaults to stop the inverter from charging the battery. I'd like to be able to write an automation that after a HA restart, it sets Fronius to defaults to avoid it charging outside of the favourable window. Thank you!

atbore-phx, 2024-10-03:
Hi @andreapalisca,

I suggest to activate those two options in the add-on configuration panel to enable an auto restart when HA will be back online.
[image]

When you are outside the working windows and you want to create your automation you can execute sham via it's standalone binary available under the release section of this repository.
The command option is `sbam configure` you have a flag for defaults (-d)

closed 2024-10-03 by atbore-phx

andreapalisca, 2024-10-03:
Thank you for your help @atbore-phx. It would be great to have these commands available somehow also in HACS Add-on installation mode.