atbore-phx/sbam #26: Charge logic question

andreapalisca, 2024-10-03:
Hi @atbore-phx,

I've been using SBAM for a few weeks and really appreciate your work - thank you. 

I have a question on the charge logic. Due to bad weather, the forecast for today was very low solar production. Even so, SBAM decided not to fully charge my 16 kWh battery in the nighttime hours and only went from 8% to 27% charge. Why is that? I would expect SBAM to charge almost to 100% in order to get through the next day. Here is the log, a graph of my charge state, and settings:

`{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Forecast Solar Power for 3/10/2024: 932 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Battery Capacity to charge: 13801 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Battery Capacity Max: 16588 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"your Daily consumption is:18000 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Net Forecast Power is not enough: -17067.300000 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Battery Capacity is not enough: 2786.784000 W"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Setting Fronius Storage Force Charge at 15%"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Reading register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Reading register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Reading register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Reading register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Reading register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Writing register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Setting Fronius Storage Force Charge done."}`

[image]

[image]

Thank you!

andreapalisca, 2024-10-03:
I think I understood my issue: It is not charging more than 15% because I set max_charge=2500. I understood the description of this setting to mean "maximum draw in W from the grid" (i.e. do not draw more than 2500 W from the grid to charge the battery) instead you mean "maximum amount of battery to charge using grid power" (i.e. do not charge the battery more than x Wh using grid power). 

I think I can fix this by setting max_power = 16588. I will try tonight...

atbore-phx, 2024-10-03:
Hi @andreapalisca,
`max_charge` is the max amount of energy you draw from the grid during the charge event so for eg. If you set it to 2500W and sbam decides to charge (negative amount of power needed) the battery will be charged at 2.5KWh so in 1 hour sbam will refill your battery with an additional 15%.

Sbam makes this calculation every hour in your crontab.

The maximum limit for `max_charge` is the max amount of power you have subscribed with your electricity provider.
I suggest not setting it to the max but just leave some KW for your home consumption to avoid a grid disconnection or penalty if you go over the provider threshold.

andreapalisca, 2024-10-03:
Hi @atbore-phx, if that is the case, to charge 15% of 16588 Wh in one hour, it should have charged at 2488 W, instead it was charging only at 470 W:

[image]

In fact, over the whole window, it went only from 8 to 27%, so 19% in seven hours.

atbore-phx, 2024-10-03:
> In fact, over the whole window, it went only from 8 to 27%, so 19% in seven hours.  
> 
Yes I agree, it's not the normal behavior.
Send me please the full crontab logs every hour from midnight to 6am.

andreapalisca, 2024-10-03:
This is all I have from last night (rest got deleted): 

`{"level":"info","ts":"2024-10-03T03:00:01+02:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Forecast Solar Power for 3/10/2024: 932 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Battery Capacity to charge: 13353 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Battery Capacity Max: 16588 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"your Daily consumption is:18000 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Net Forecast Power is not enough: -17067.300000 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Battery Capacity is not enough: 3234.660000 W"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Setting Fronius Storage Force Charge at 15%"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Reading register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Reading register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Reading register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Reading register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Reading register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Writing register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T04:00:03+02:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Forecast Solar Power for 3/10/2024: 932 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Battery Capacity to charge: 12905 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Battery Capacity Max: 16588 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"your Daily consumption is:18000 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Net Forecast Power is not enough: -17067.850000 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Battery Capacity is not enough: 3682.536000 W"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Setting Fronius Storage Force Charge at 15%"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Reading register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Reading register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Reading register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Reading register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Reading register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Writing register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T05:00:02+02:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Forecast Solar Power for 3/10/2024: 883 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Battery Capacity to charge: 12474 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Battery Capacity Max: 16588 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"your Daily consumption is:18000 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Net Forecast Power is not enough: -17116.900000 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Battery Capacity is not enough: 4113.824000 W"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Setting Fronius Storage Force Charge at 15%"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Reading register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Reading register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Reading register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Reading register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Reading register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Writing register: 40356 ; value: 64036"}
{"level":"info","ts":"2024-10-03T06:00:03+02:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Setting Fronius Storage Defaults start..."}
[... cut ...]
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Writing register: 40356 ; value: 10000"}
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-10-03T06:55:00+02:00","msg":"Setting Fronius Modbus Defaults done."}`

I can send full log tomorrow.... 

Thank you!

andreapalisca, 2024-10-04:
Hi @atbore-phx, I finally figured out what the problem was: I had not enabled the "Allow battery charging from public grid" setting in the Fronius web UI. That was limiting the battery charge current. For anyone who may have the same issue, this is where the setting is:

[image]

No issue with SBAM :-) Thanks for your help. And great work!

atbore-phx, 2024-10-25:
@andreapalisca i'll change the doc adding the warning `Allow battery charging from public grid` this weekend making a new release

atbore-phx, 2024-10-26:
https://github.com/atbore-phx/sbam/releases/tag/v1.3.7

closed 2024-10-26 by atbore-phx