atbore-phx/sbam #39: Charging does not start

Johnnexto, 2024-11-12:
Thank you for your good work!

Battery charging does not start if net battery capacity is not enough.
Please consider to trigger charging as long as available storage is more than max_charge

Please add trigger to write initial register values at end_hr. (Or add an option to do.)
To stop forced loading from grid at expensive hours. (even if power to charge is not achieved)
[image]

Two logs below, and solar web consumption.

Thanks 

{"level":"info","ts":"2024-11-12T01:11:01+01:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2024-11-12T01:11:01+01:00","msg":"Forecast Solar Power for 12/11/2024: 29880 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Starting Calculate PV production for solar System 1"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Forecast Solar Power for 12/11/2024: 9299 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Total Forecast Solar Power for 12/11/2024: 39179 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Battery Capacity to charge: 20536 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Battery Capacity Max: 33177 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"your Daily consumption is:52000 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Net Forecast Power is not enough: -12820.350000 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Battery Capacity is not enough: 12640.437000 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Setting Fronius Storage Defaults start..."}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Reading register: 40349 ; value: 0; default:"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Reading register: 40356 ; value: 10000; default:"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Writing register: 40349 ; value: 0"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Writing register: 40356 ; value: 10000"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Setting Fronius Modbus Defaults done."}

{"level":"info","ts":"2024-11-12T02:11:01+01:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2024-11-12T02:11:01+01:00","msg":"Forecast Solar Power for 12/11/2024: 30050 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Starting Calculate PV production for solar System 1"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Forecast Solar Power for 12/11/2024: 9523 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Total Forecast Solar Power for 12/11/2024: 39573 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Battery Capacity to charge: 22593 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Battery Capacity Max: 33177 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"your Daily consumption is:52000 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Net Forecast Power is not enough: -12426.200000 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Battery Capacity is not enough: 10583.463000 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Setting Fronius Storage Force Charge at 5%"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Reading register: 40349 ; value: 0; default:"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Reading register: 40356 ; value: 10000; default:"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Writing register: 40356 ; value: 65036"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Setting Fronius Storage Force Charge done."}

atbore-phx, 2024-11-12:
Hi @Johnnexto!

Thanks for using SBAM!

frm the doc:
`max_charge`: Maximum amount of power required from the electricity network to charge the battery in W (Default: 3500)

For more details about `max_charge` parameter, please check this issue: https://github.com/atbore-phx/sbam/issues/26#issuecomment-2391048862.

> Please consider triggering charging as long as available storage is more than max_charge.

I think the correct parameter is `pw_batt_reserve`: Minimum battery capacity to maintain in Wh (Default: 4000, which means 4kWh).

> Battery charging does not start if net battery capacity is not enough.

In the first iteration on `2024-11-12T01:11:01+01:00`, the battery was not charged:

```
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Net Forecast Power is not enough: -12820.350000 Wh"}
{"level":"info","ts":"2024-11-12T01:11:02+01:00","msg":"Battery Capacity is not enough: 12640.437000 Wh"}
```
This is not a bug; the reason is the approximation. You need just 180W (which is too small) to fill the gap: 12820 - 12640. This function calculates the percentage to add in int16: https://github.com/atbore-phx/sbam/blob/4332a654391815ff59961fb81ad24ef645aec7ac/pkg/fronius/schedule.go#L50. So, 180W in your case is 0.54%, which approximates to 0% in int16, so it was not charged.

Anyway, I need to be more explicit in the logs to avoid confusion. I take the point.

In the second iteration, you charged 5% (10583.463 - 12426.2 is approximately 5% of 33177, your full battery capacity):

```
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Net Forecast Power is not enough: -12426.200000 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Battery Capacity is not enough: 10583.463000 Wh"}
{"level":"info","ts":"2024-11-12T02:11:02+01:00","msg":"Setting Fronius Storage Force Charge at 5%"}
```

> To reset all Modbus values: Disable Modbus in Settings and re-enable. This resets all values to default.

I'll try to be more explicit in the documentation: https://github.com/atbore-phx/sbam/blob/main/home-assistant/addons/sbam/DOCS.md. 

Just enable this in your configuration:

[image]

Defaults: At the end of the crontab cycle, reconfigure the Fronius inverter to default (automatic management). The end of the crontab cycle is end_hr - 5 min. For example, if you have these parameters:

start_hr: Start time of the advantageous network operator rate (default 00:00)
end_hr: End time of the advantageous network operator rate (default 06:00)
crontab: Crontab to run sbam (default: 00 00-05 * * *)
SBAM evaluates whether to charge or not between 12 AM (midnight) and 5 AM. SBAM will set the defaults at 5:55 AM.

I'll try to be more explicit in the documentation to avoid confusion. Thanks!

Johnnexto, 2024-11-12:
Thank you very much for taking time to explain, this make perfect sense. I totally misread the max_charge (confused it with the "Max. Charging Power from AC" set in Battery mgmt on inverter.) and the defaults in the config section is clearly described in the documentation.

atbore-phx, 2024-11-12:
@Johnnexto,

Thanks a lot for your questions. Improving the documentation is mandatory, so your feedback is welcome 👍.

Regarding the definitions of `max_charge `and `defaults`, I will leave this issue open and address it in the next release. I will rewrite these two points more clearly in the documentation.

Best !

closed 2024-11-17 by atbore-phx