atbore-phx/sbam #48: Integrate Cheapest Energy Hours Macro

greteOne, 2024-12-11:
Hello everyone,

not an issue but unfortunately the best place for this idea:
is it possible to include the following macro via a helper entity, for example, so that dynamic charging according to the cheapest energy price would be possible? The combination of the add-on with the macro would be a gamechanger in my eyes.

https://github.com/TheFes/cheapest-energy-hours

Kind regards

atbore-phx, 2024-12-15:
Hi @greteOne,  

Thanks for your feature request!  

This feature could automatically define `start_hr` and `end_hr` in the SBAM configuration. While I’m not familiar with this extension, if someone wants to create a PR to implement it, I’m fine with that.  

### Possible Implementation  
- Introduce a boolean variable, `auto_setup_time_range` (default: `false`).  
  - If `true`, `start_hr` and `end_hr` will be defined via the output of the macro (via API request ?).  
  - If `false`, they will be defined manually as usual.  

I’ve labeled this as a **Good First Issue** for anyone interested in contributing.

greteOne, 2025-01-22:
think that the boolean and the start time are sufficient. 
The specification for the end time results from the maximum charging power from the grid to the battery and can be included in the macro.
What is the maximum charging power from the grid?

atbore-phx, 2025-01-22:
> What is the maximum charging power from the grid?

the parameter is 
`max_charge: Maximum amount of power required from the electricity network to charge the battery in W (Default: 3500)`

You can configure this parameter to limit the maximum amount of energy used to charge your battery. This is particularly useful for preventing your electricity consumption from exceeding the capacity allowed by your contract, thereby avoiding shutdowns or penalties from your provider.

For example, if your electricity contract allows for a maximum of 6 kW, and your household appliances are already consuming 4 kW, then attempting to charge your battery with 3 kW would exceed your limit (4 kW + 3 kW > 6 kW). This could lead to your system being shut down or incurring additional charges.

By setting the maximum charging power to, say, 1.5 kW, even if your battery requires 3 kW, the system will limit the charging to 1.5 kW, ensuring you stay within your contracted limit.

atbore-phx, 2026-05-17:
related to #122 and #62 (multiple start_hr, end_hr).
this issue introduces the concepts to configure automatically via macro the convenient charge windows but it's blocked by #62 because today we support just one time slot