atbore-phx/sbam #49: Dynamic Batteries Charging

giuliob97, 2024-12-17:
Hello everyone!
First of all thanks for the add-on.
I was wondering if it is possible to charge the batteries in a dynamic way. I'll try to explain it better.
I have a energy contract in which i pay the same independently from my consumption. 
I would like to charge the batteries while my house is absorbing few kw with a formula like this: 6Kw - Total Absorption.

Is it possible?

atbore-phx, 2025-01-22:
if someone wants to create a PR to implement it, I’m fine with that.

travellingkiwi, 2025-08-30:
My Fronius inverter (A 6kW Gen24+ with BYD batteries) does this automatically. Are there some Fronius inverters that don't do this? Or is it a config issue?