atbore-phx/sbam #60: you have exceeded your free daily limit, too many Request to the forecast API

Mense1978, 2025-02-09:
My configuration is as follows:

start_hr:4:00
end_hr:9:00
crontab:1 4-9 * * *
batt_reserve_start_hr: 4:00
batt_reserve_end_hr: 07:59

Every time it runs at 04:00, I get this error in the log:
you have exceeded your free daily limit, too many Request to the forecast API"

It only runs until 09:00, when it shows this.

"The current time is outside the range defined by start_hr and end_hr.: 04:00 <= t <= 9:00"}

Can you please help me?

Thanks

atbore-phx, 2025-02-09:
hi @Mense1978,

We previously discussed this issue here: [GitHub Issue #40](https://github.com/atbore-phx/sbam/issues/40).

In summary, Solcast has a limit of 10 API calls per day for free accounts. Each site/roof requires one API call, and the limit resets at midnight **UTC** time zone.

closed 2025-02-09 by atbore-phx

Mense1978, 2025-02-09:
I know, but I am not using 10 API calls, I start calling at 4 in the morning, getting that error.

atbore-phx, 2025-02-09:
@Mense1978 You need to check with Solcast, as this message comes from their API.

Also, verify the number of installations linked to your account. If you have more than one, your API limit is divided accordingly.

For example, with 2 installations, the limit would be 5 API calls per day.

Also, check your timezone in relation to midnight UTC (universal time coordinated) to avoid exceeding the reset interval.