atbore-phx/sbam #61: Solar forecast caching

travellingkiwi, 2025-03-08:
Currently the free solar forecast is only able to be queried 10 times a day. If you then set the resolution of the sbam running to more than once an hour, you generally will run out of queries. When queries expire, the solar forecast is calculated as 0, and... battery starts to charge...

It's unlikely the forecast will actually change much during the run period. So... caching the results would be a good thing.

atbore-phx, 2025-03-08:
Hi @travellingkiwi,
Thank you for the suggestion! I completely agree that caching the last execution's data would be a great improvement, especially given the query limitations of the SolCast API. This will help ensure more stable and accurate forecasts when the API limit is reached.

I’ll make sure to implement this in the next version of the application. If anyone is interested in contributing to the development of this feature, they are more than welcome to help!

Thanks again for the valuable input!
Best

gnana997, 2025-03-08:
Hi @atbore-phx , Can I try to work on this ?

atbore-phx, 2025-03-08:
Hi @gnana997, 
Only if your motorbike is a Ducati! :)
Of course, you can. I'm really glad if you could help.
How would you like to implement this cache?
Thanks for your help!"

felsnerino, 2025-05-26:
Hi - wouldn't it be better to just reuse the values available through the other integration for Solcast? At the end several automations want to access the values: https://github.com/BJReplay/ha-solcast-solar
Sorry for the German: 

[image]

atbore-phx, 2025-08-30:
the cache feature has been implemented by @travellingkiwi  in [v1.5.0](https://github.com/atbore-phx/sbam/releases/tag/v1.5.0).
well done @travellingkiwi thanks for your contribution !!!! 🚀

closed 2025-08-30 by atbore-phx