atbore-phx/sbam #65: sbam doesn't charge at selected rate

travellingkiwi, 2025-03-18:
For some reason when sbam charges the battery for off-peak, it elects to use an artificially low charge rate (Between 1% and 21% usually) which often results in the battery not reaching the minimum specified charge before the off-peak time ending forces the battery charging to stop.

e.g. this morning
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"Battery Capacity to charge: 10243 Wh"}
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"your Daily consumption is:30000 Wh"}
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"battery 3580.416000 Wh < reserve 4000.000000 Wh"}
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"Setting Fronius Storage Force Charge at 3%"}
{"level":"info","ts":"2025-03-19T06:00:00+13:00","msg":"Setting Fronius Storage Force Charge done."}

A 3% charge rate is insufficient to reach the minimum

If I artificially force it (Bumping the minimum reserver to 10000 and allowing 15 minutes only with a max charge of 5000) it till only attempts charging at 36%
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"Battery Capacity to charge: 10008 Wh"}
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"your Daily consumption is:30000 Wh"}
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"battery 3815.424000 Wh < reserve 10000.000000 Wh"}
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"Setting Fronius Storage Force Charge at 36%"}
{"level":"info","ts":"2025-03-19T06:45:00+13:00","msg":"Setting Fronius Storage Force Charge done."}

Before setting the normal discharge mode 10 minutes later

{"level":"info","ts":"2025-03-19T06:55:00+13:00","msg":"Setting Fronius Storage Defaults start..."}
{"level":"info","ts":"2025-03-19T06:55:00+13:00","msg":"Setting Fronius Modbus Defaults done."}

travellingkiwi, 2025-04-01:
Today it's even worse. Till now it's only been the minimum charge that was affected. Today is a particularly low day for solar (8kWh predicted). And still sbam refused to charge the batteries more than 3kWh. 
Clearly something was broken in one of the recent updates. Possibly the hysteresis update? If my batteries need to charge I'd prefer they did it limited by the max charge rate, not an artificially low percentage of it.

```
{"level":"info","ts":"2025-04-02T04:00:00+13:00","msg":"Setting Fronius Storage Force Charge at 36%"}
{"level":"info","ts":"2025-04-02T04:00:00+13:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Forecast Solar Power for 2/4/2025: 8808 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Total Forecast Solar Power for 2/4/2025: 8808 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Battery Capacity to charge: 11017 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"your Daily consumption is:20000 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -8385.628000 Wh"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Setting Fronius Storage Force Charge at 36%"}
{"level":"info","ts":"2025-04-02T05:00:00+13:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Forecast Solar Power for 2/4/2025: 8808 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Total Forecast Solar Power for 2/4/2025: 8808 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Battery Capacity to charge: 10492 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"your Daily consumption is:20000 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -7860.316000 Wh"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Setting Fronius Storage Force Charge at 36%"}
{"level":"info","ts":"2025-04-02T06:00:00+13:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-04-02T06:55:00+13:00","msg":"Setting Fronius Storage Defaults start..."}
{"level":"info","ts":"2025-04-02T06:55:00+13:00","msg":"Setting Fronius Modbus Defaults done."} 
```

atbore-phx, 2025-04-01:
Hi @travellingkiwi, sorry for the late reply, busy week at work ;)

Looking at your logs, SBAM triggers the charge, but the battery level doesn’t increase accordingly. This seems similar to an issue we previously encountered with @andreapalisca: https://github.com/atbore-phx/sbam/issues/26#issuecomment-2393334343.

Could you check if the "charging from the grid" function is enabled?
Try also to trigger a charge manually directly from your fronius inverter.

Please also share your full configuration (excluding your site URL) and activate debug mode.

atbore-phx, 2025-04-01:
By the way the hysteresis mode can be excluded by just setting the defaults (0) for those parameters:
pw_lwt: The hysteresis logic lower threshold offset in Wh to stop charging (Default: 0)
pw_upt: The hysteresis logic upper threshold offset in Wh to start charging (Default: 0)

travellingkiwi, 2025-04-01:
Possibly related to issue #26 

My Gen24 (6kW with BYD battery) does have slightly different options though. Possibly this was reset by a recent inverter firmware update? Because it used to work fine.

[image]

Will test with (re)enabling the option.

Query. Is it possible for that setting to be managed by sbam? To enable that when grid charging. And disabling it when grid charging stops?

atbore-phx, 2025-04-02:
I have the same setup, yes the UI is quite different than the issue mentioned but yes the right setting is to enable the charge from the public grid.

Looking at your picture I have the Soc limit in auto but I don't think it's relevant.

I have never experienced those changes after a firmware update, so I would say no. 

Regarding the second question, SBAM can't change the config options because the HTTP API is read-only by design for the Fronius systems.

SBAM uses only modbus TCP to trigger a charge and this option seems not available.

Closing the issue
Have a nice day with SBAM !

closed 2025-04-02 by atbore-phx