atbore-phx/sbam #66: Minimum charge capacity is ignored when net power required is < minimum capacity

travellingkiwi, 2025-05-19:
Settings
Min Charge Level 5000Wh
Daily Usage 33000Wh
Current Charge Level 2.9kWh
Forecast 30000Wh

When the daily usage-forecast is low (e.g. as above) then sbam will only charge to the live required to supply 33000Wh. It will ignore the 5kWh min charge level (Which is required to supply electricity through the peak hours of the morning before the solar kicks in)

Logs

 {"level":"info","ts":"2025-05-20T02:00:01+12:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -1552.974000 Wh"}
{"level":"info","ts":"2025-05-20T02:00:01+12:00","msg":"Setting Fronius Storage Force Charge at 11%"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Reading register: 40349 ; value: 0; default:"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Reading register: 40356 ; value: 10000; default:"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Writing register: 40356 ; value: 64436"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"debug","ts":"2025-05-20T02:00:01+12:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"info","ts":"2025-05-20T02:00:01+12:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Total Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Battery Capacity to charge: 11155 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"your Daily consumption is:33000 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -253.518000 Wh"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Setting Fronius Storage Force Charge at 1%"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Reading register: 40349 ; value: 2; default:"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Reading register: 40356 ; value: 64436; default:"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"debug","ts":"2025-05-20T03:00:01+12:00","msg":"Writing register: 40356 ; value: 65436"}
{"level":"info","ts":"2025-05-20T03:00:01+12:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Total Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Battery Capacity to charge: 11031 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"your Daily consumption is:33000 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -129.102000 Wh"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Setting Fronius Storage Force Charge at 0%"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"percent of charging is <1%, skipping Force Charge and set defaults."}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Setting Fronius Storage Defaults start..."}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Reading register: 40349 ; value: 2; default:"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Reading register: 40356 ; value: 65436; default:"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Writing register: 40349 ; value: 0"}
{"level":"debug","ts":"2025-05-20T04:00:01+12:00","msg":"Writing register: 40356 ; value: 10000"}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Setting Fronius Modbus Defaults done."}
{"level":"info","ts":"2025-05-20T04:00:01+12:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"Total Forecast Solar Power for 20/5/2025: 30078 Wh"}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"Battery Capacity to charge: 11791 Wh"}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"Battery Capacity Max: 13824 Wh"}
{"level":"info","ts":"2025-05-20T05:00:00+12:00","msg":"your Daily consumption is:33000 Wh"}
[... 19 lines cut ...]
{"level":"info","ts":"2025-05-20T06:00:00+12:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -163.354000 Wh"}
{"level":"info","ts":"2025-05-20T06:00:00+12:00","msg":"Setting Fronius Storage Force Charge at 1%"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Reading register: 40349 ; value: 2; default:"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Reading register: 40356 ; value: 64936; default:"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Writing register: 40356 ; value: 65436"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T06:00:00+12:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2025-05-20T06:00:00+12:00","msg":"Setting Fronius Storage Force Charge done."}
{"level":"info","ts":"2025-05-20T06:55:00+12:00","msg":"Setting Fronius Storage Defaults start..."}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Reading register: 40349 ; value: 2; default:"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Reading register: 40356 ; value: 65436; default:"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Reading register: 40357 ; value: 10000; default:"}
[... cut ...]
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Writing register: 40349 ; value: 0"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Writing register: 40356 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"debug","ts":"2025-05-20T06:55:00+12:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"info","ts":"2025-05-20T06:55:00+12:00","msg":"Setting Fronius Modbus Defaults done."}

travellingkiwi, 2025-05-19:
This would appear to be because of a bug in fronius/schedule.go (func SetFroniusChargeBatteryMode) where a case statement is used to calculate the charge percentage. So if the Net Power Required is > 0, the minimum charge level is ignored

```
        switch {
        case pw_batt2charge == 0: // battery 100% => do not charge
                u.Log.Info("Battery is full charged")
        case pw_net < -1*pw_lwt && forecast_charge_enabled: // net power is not enough => charge
                u.Log.Infof("Net Power (actual battery power + Net solar power) is not enough: %f Wh", pw_net)
                ch_pc = SetChargePower(pw_batt_max, -1*pw_net+pw_upt, max_charge)
        case pw_batt_reserve_net < -1*pw_lwt && batt_reserve_charge_enabled: // battery is less than reserve and reserve battery is time enabled => charge
                u.Log.Infof("battery %f Wh < reserve %f Wh", pw_batt+pw_upt, pw_batt_reserve)
                ch_pc = SetChargePower(pw_batt_max, pw_batt_reserve-pw_batt, max_charge)
        default: // net power is enough => do not charge
                u.Log.Infof("Net Power (actual battery power + Net solar power) is enough: %f Wh", pw_net)
        }
```

This should probably be written more like (pseudo code - I may get some time to code & test this later today. Maybe)

```
  fc_ch_pc=0
  min_ch_pc=0
  if(pw_batt2charge==0) {
    u.Log.info("Battery is fully charged")
  } else if(pw_net < -1*pw_lwt && forecast_charge_enabled) {
    u.Log.Infof("Net Power (actual battery power + Net solar power) is not enough: %f Wh", pw_net)
    fc_ch_pc = SetChargePower(pw_batt_max, -1*pw_net+pw_upt, max_charge)
  }
  if(pw_batt_reserve_net < -1*pw_lwt && batt_reserve_charge_enabled) {
    u.Log.Infof("battery %f Wh < reserve %f Wh", pw_batt+pw_upt, pw_batt_reserve)
    min_ch_pc = SetChargePower(pw_batt_max, pw_batt_reserve-pw_batt, max_charge)
  }
  ch_pc=max(min_ch_pc, fc_ch_pc)
  if(fc_ch_pc!=0) {
    u.Log.Infof("Net Power (actual battery power + Net solar power) is not enough: %f Wh", pw_net)
  }
  if(min_ch_pc!=0) {
    u.Log.Info("Below minimum reserve")
  }
  if(ch_pc!=0) {
    u.log.Info("Calculated ch_pc required %f")
  }
```

Sorry, not sure at first glance what units ch_pc is in... But it should be the max of either required charge for forecast or required charge for reserve, and not simply the required charge and only reserve charge if required charge is <=0

atbore-phx, 2025-05-19:
Hi @travellingkiwi ,
Could you please provide the full list of your settings? (You can omit url, apikey and fronius_ip)

travellingkiwi, 2025-05-19:
```
fronius_ip: 192.168.0.88
start_hr: "01:01"
end_hr: "07:00"
crontab: 0 02-06 * * *
pw_consumption: 33000
max_charge: 5000
pw_lwt: "0"
pw_upt: "0"
pw_batt_reserve: 5000
batt_reserve_start_hr: "02:00"
batt_reserve_end_hr: "07:00"
defaults: "true"
reset: false
debug: true
```

atbore-phx, 2025-05-19:
In my initial analysis, I don't believe this is a bug.
The live forecast charging case is evaluated before the battery reserve case (so its priority is bigger).
In your logs, the live forecast is always triggered because the battery charge is close to the requested threshold.

I suggest adding a bit of hysteresis to avoid repeated triggering.

You could try setting:

pw_lwt = 1000
pw_upt = 1000
It should limit your behavior because the live forecast case will be satisfied in the latest cycles and the battery reserve case will be evaluated.

atbore-phx, 2025-05-19:
In my latest analysis, the charging logic prioritizes the live forecast over the battery reserve.

This makes sense, as the main goal of SBAM is to use the forecast.
However, I understand the concern—when we're close to the threshold and the PV installation produces less power in the morning, this can be problematic.

Perhaps we could add a parameter to choose the logic order, e.g., forecast-first or battery-reserve-first, to give more control over the priority.

CHODEUS, 2025-09-05:
I feel I have the same issue. Today is a low solar production day as it is raining and there’s no way my battery is going to be at 100% before the end of the off-peak power which runs from 0900-1500. 
It’s currently only 56% and 11:30am. 

Or have I set this up wrong?
My crontab is set like that to maximise my 10 API calls during the off-peak power window

`fronius_ip: 192.168.2.229
start_hr: "09:00"
end_hr: "15:00"
crontab: "*/37 9-14 * * *"
pw_consumption: 16000
max_charge: 3500
pw_lwt: "0"
pw_upt: "0"
pw_batt_reserve: 1500
batt_reserve_start_hr: "09:00"
batt_reserve_end_hr: "15:00"
defaults: true
reset: true
debug: true
cache_forecast: true
cache_time: 7200
cache_file_prefix: cached_forecast`

`(*Cron).startJob.func1\n\t/home/runner/go/pkg/mod/github.com/robfig/cron/v3@v3.0.1/cron.go:312"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Info: Cache File 'cached_forecast.0' does not exist - fallthough to download"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Falling through to URL read for forecast"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Cachefile cached_forecast.0 written successfully"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Starting Calculate PV production for solar System 0"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Forecast Solar Power for 5/9/2025: 4019 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Total Forecast Solar Power for 5/9/2025: 4019 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Battery Capacity to charge: 9231 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Battery Capacity Max: 19353 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"your Daily consumption is:16000 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Net Power (actual battery power + Net solar power) is not enough: -1858.831000 Wh"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Setting Fronius Storage Force Charge at 9%"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Reading register: 40356 ; value: 10000; default:"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Reading register: 40357 ; value: 10000; default:"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Reading register: 40351 ; value: 0; default:"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Reading register: 40361 ; value: 1; default:"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Reading register: 40349 ; value: 0; default:"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Writing register: 40351 ; value: 0"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Writing register: 40361 ; value: 1"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Writing register: 40349 ; value: 2"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Writing register: 40356 ; value: 64636"}
{"level":"debug","ts":"2025-09-05T11:00:00+08:00","msg":"Writing register: 40357 ; value: 10000"}
{"level":"info","ts":"2025-09-05T11:00:00+08:00","msg":"Setting Fronius Storage Force Charge done."}`

closed 2025-09-05 by atbore-phx